goodrobots/MAVCesium #2: 16.04 mate throws unexpected keyword argument 'root_path'

brizey02, 2016-12-17:
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 491, in process_stdin
    fn(args[1:])
  File "/usr/local/bin/mavproxy.py", line 319, in cmd_module
    load_module(args[1])
  File "/usr/local/bin/mavproxy.py", line 277, in load_module
    module = m.init(mpstate)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_cesium/__init__.py", line 266, in init
    return CesiumModule(mpstate)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_cesium/__init__.py", line 72, in __init__
    self.run_web_server()
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_cesium/__init__.py", line 97, in run_web_server
    from app import cesium_web_server
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_cesium/app/cesium_web_server.py", line 26, in <module>
TypeError: __init__() got an unexpected keyword argument 'root_path'

SamuelDudley, 2016-12-18:
Flask is the python webserver that MAVCesium uses and the version you have installed may not support the root_path argument.

Can you please tell me what version of Flask you have installed / are using?

brizey02, 2016-12-18:
Name: Flask
Version: 0.10.1
Summary: A microframework based on Werkzeug, Jinja2 and good intentions
Home-page: http://github.com/mitsuhiko/flask/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD
Location: /usr/lib/python2.7/dist-packages
Requires: Werkzeug, Jinja2, itsdangerous

SamuelDudley, 2016-12-18:
Looking at the Flask docs the minimum version required is 0.11 so I'll update the requirements.txt file to reflect this.
Current version installed from pip is 0.11.1 so try the following and see if the issue is fixed:

pip install Flask --upgrade

brizey02, 2016-12-18:
(Opcode -1) on websocket data now.

SamuelDudley, 2016-12-18:
This will most likely be linked to autobahn rather than flask. Did you install autobahn[twisted] via pip?

brizey02, 2016-12-18:
yes same problem one frame per chrome debug then nothing.

SamuelDudley, 2016-12-19:
So it's sending the mavlink defines (first thing that occurs when you start a new connection) but isn't streaming continuous data past that point?
I'm just trying to work out if this is browser side or an issue with the websocket server and trying to come up with a way to help you debug this.

brizey02, 2016-12-21:
can you share the script you use to automate the flight loop, maybe its something with my sitl not working correctly?

SamuelDudley, 2017-02-13:
I was having a bit more of a think about this and it might be a Cross-origin Resource Sharing (CORS) issue unique to your setup. Is there any info in the chrome debug to provide any hints?

closed 2017-09-24 by brizey02