goodrobots/MAVCesium #3: 14.04 ImportError: No module named twisted.websocket

brizey02, 2016-12-17:
STABILIZE> module load cesium
STABILIZE> Traceback (most recent call last):
  File "mavproxy.py", line 276, in load_module
    m = import_package(modpath)
  File "mavproxy.py", line 406, in import_package
    mod = __import__(name)
  File "build/bdist.linux-x86_64/egg/MAVProxy/modules/mavproxy_cesium/__init__.py", line 15, in <module>
ImportError: No module named twisted.websocket

Traceback (most recent call last):
  File "mavproxy.py", line 276, in load_module
    m = import_package(modpath)
  File "mavproxy.py", line 406, in import_package
    mod = __import__(name)
ImportError: No module named cesium

Failed to load module: No module named cesium. Use 'set moddebug 3' in the MAVProxy console to enable traceback

[1]+  Stopped                 python mavproxy.py --master=tcp:127.0.0.1:5760 --c

SamuelDudley, 2016-12-18:
Hi, it looks like you don't have the all the requirements installed on your machine and it's crashing out looking for 
autobahn[twisted].

Depending on your python install you may or may not need sudo for the following:

sudo pip install autobahn[twisted] 

Let me know if that fixes it.

brizey02, 2016-12-18:
zero@zero:~$ dpkg -s python-autobahn
Package: python-autobahn
Status: install ok installed
Priority: optional
Section: python
Installed-Size: 885
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Version: 0.10.3+dfsg1-5
Depends: python-concurrent.futures (>= 2.1.5), python-lz4 (>= 0.2.1), python-msgpack (>= 0.4.0), python-six (>= 1.6.1), python-snappy (>= 0.5), python-trollius (>= 0.1.2), python-twisted (>= 11.1), python-txaio (>= 1.0.0), python-zope.interface (>= 3.6), python:any (<< 2.8), python:any (>= 2.7.5-5~)
Description: WebSocket client and server library, WAMP framework - Python 2.x
 Autobahn|Python is a networking library that is part of the Autobahn project
 and provides implementations of:
  * The WebSocket Protocol
  * The Web Application Messaging Protocol (WAMP)
 for Twisted and asyncio, on Python 2 & 3 and for writing servers and clients.
 .
 WebSocket allows bidirectional real-time messaging on the Web and WAMP adds
 asynchronous Remote Procedure Calls and Publish & Subscribe on top of
 WebSocket.
 .
 This package contains the Python 2.x module.
Original-Maintainer: PKG OpenStack <openstack-devel@lists.alioth.debian.org>
Homepage: http://autobahn.ws/python

SamuelDudley, 2016-12-18:
Just to confirm how did you install autobahn? It looks like you have used apt-get... I have only used pip to install as per my last comment. If you have not used pip are you able to give that a try? 

sudo pip install autobahn[twisted]

SamuelDudley, 2017-02-13:
Closing this issue as the dependencies have been identified in #2 and fixed.

closed 2017-02-13 by SamuelDudley