goodrobots/MAVCesium #4: no webpage at localhost:5000 , how to troubleshoot ?

AndKe, 2017-02-12:
I did install as expected using git clone , and submodules.
I see no errors, but nor do I see anything at port 5000.. :(

AndKe, 2017-02-12:
apparently it works once mavlink data is received, not before..

SamuelDudley, 2017-02-13:
Hi,
Glad it worked for you.
I'll make an issue for this, as it would be nice to have the system give some feedback prior to mavlink data.

Let me know if you run into other issues and ill do my best to help out.
Thanks,
Sam

AndKe, 2017-02-13:
Hi, thank you for the software, I am happy to see you are still working on it.  (was unsure due to long time since last commit)
Until the server starts before mavlink data, consider updating the readme.md with  information about it, following the readme I expected to see something at port 5000 and assumed something failed.

SamuelDudley, 2017-02-13:
Updated the readme so that the user knows valid mavlink msg's are required prior to visualisation ee6584d496a67ebe4c5d532d1d17c9ba05e3ff37

SamuelDudley, 2017-02-20:
added a waiting for data message #9

AndKe, 2017-02-20:
on load, it would be nice if a message with the URL popped up in mavproxy console, it could be clickable, but would always he helpful to those who have not used cesium a lot yet.

SamuelDudley, 2017-02-20:
Thanks, that's a great idea! I'll see what I can do about the clickable link, but a text output will be quite simple.

AndKe, 2017-02-20:
Text output should be enough,  additionally, you could ask system to open
the URL, so it would start in  browser  right away.

AndKe, 2017-02-20:
... basically
https://www.google.no/amp/s/pythonconquerstheuniverse.wordpress.com/2010/10/16/how-to-open-a-web-browser-from-python/amp/

SamuelDudley, 2017-02-20:
Hi,
Thanks for the links. For the moment I have the module printing a message in the MAVProxy console and auto loading a tab. If the MAVProxy [PR](https://github.com/ArduPilot/MAVProxy/pull/381) is accepted I will remove the auto load feature from MAVCesium and the user can easily double click the link from the MAVProxy console window.

SamuelDudley, 2017-02-20:
Thanks again for the input. I'll have a look at implementing some of the trickier HUD / UI bits after I have the configuration control worked out... Feel free to keep adding ideas / requests to the enhancements issue.
Cheers,
Sam

closed 2017-02-21 by SamuelDudley

SamuelDudley, 2017-03-15:
https://github.com/ArduPilot/MAVProxy/pull/381was merged today adding support for click-able URLs in the MAVProxy console.