goodrobots/MAVCesium #6: Feature requests to improve usefullness

AndKe, 2017-02-13:
Please consider adding :
-wind speed & direction. (numerical with an arrow ?)
-xtrack and altitude error
heading, bearing , and next few waypoints
in the the mockup below heading is not very relevant, but in other views it may be.

Together, viewing turns with targets (WP right in the map)  - would make it easy to evaluate turns when tuning. 
[image]

SamuelDudley, 2017-02-13:
Hi, 
Thanks for the suggestions. Re the addition of the next few waypoints, can you not see wp's on the display (blue triangles, currently without labels) as per the [images](https://github.com/SamuelDudley/MAVCesium/blob/gh-pages/screenshots/CMAC_HUD.png) in the readme?

AndKe, 2017-02-14:
I must say I played back a tlog from previous mission, I've seen before that mavproxy fail to get some/most waypoints when playing back tlog (that contain mission upload) rather than actually flying - I do not know whatever mavcesium reads mavproxy WP, or reads the stream(or requests) them by itself, but playback must be the reason..

SamuelDudley, 2017-02-14:
By design mavcesium will keep in sync with the wp's that are stored in mavproxy. So if you can see them on the mavproxy map you should also see them in mavcesium... I haven't tried visualising log playback with mavcesium running. I'll add it to the list of things to try!

SamuelDudley, 2017-02-20:
I have added the wind and nav performance info to the default hud via #7. This leads me to #8, because one hud will not be suitable for all vehicles / use applications.

SamuelDudley, 2017-03-26:
I'm quite a fan of this HUD: http://gallinazo.flightgear.org/uas/autopilot-visualization-flight-track/
I would like to implement an ardupilot supported version of this HUD with two back ends:
1. HTML5
2. OpenCV

AndKe, 2017-03-27:
nope - that sounds too awesome to be true :)
... cool ! :)

SamuelDudley, 2017-06-24:
A quote from @SamPes capturing some really good ideas
> I really like UGC "3D" interface, but it doesn't support QP well enough. Ideally I'd use Mavcesium :grinning: I like the idea of showing a locally run 3D "flight simulation" GCS, that is then augmented with real (time) data, for good 3D situational awareness, regardless of what comms/bandwidth is available. (In fact the simulation could compute the expected position on comms loss even) I'd also like to have a time slider across the bottom of the screen, that allows me to slide left to see the log (and watch the 3D simulation scroll back into the past using log data), then click the middle to see real time, then slide right to see the current flight plan (into the future with 3D simulation). Then bring up tuning graphs with physical representations on the 3D model of desired versus achieved control inputs etc. It would also only display context sensitive data for the task at hand. Then finally only have one 3D map screen, for both mission control and planning, by using a scrolling tab system on either side of the screen to bring up whatever settings widget to change WP, comms, FPV video, image/thumbnail gallery and parameters etc that then compresses the map view, without loosing situational awareness because I "happen" to be on another screen trying to change something in flight. Sort of a hybrid 3D glass cockpit GCS using mouse/touchscreen and without knobs. :grinning: