goodrobots/MAVCesium #18: Make synthetic camera default?

fnoop, 2017-05-14:
When you first load the webpage it looks like camera points down, so it's very unclear what you're looking at.  The 'synthetic' camera provides a familiar horizon.  Also, perhaps 'synthetic' might be called 'forward' to be more intuitive?

SamuelDudley, 2017-09-27:
Hi,
I have renamed the synthetic view to 'forward' with [this](https://github.com/SamuelDudley/MAVCesium/commit/453fc17a8d3a7fc68c0b0211fabc0c71d3fca5c4) commit .
Re what the default view should be. I feel it should be a user option at minimum and when I rework the javascript for the view controller I will re assess the default option.
Cheers!

fnoop, 2017-09-27:
Revisiting mavcesium again yesterday, I loaded it up and took off, and thought it was broken.  The bing map it takes off from (presume it's the flying field in Aus) is very dark and faded (overlay?) and hard to see anything, just some changing blobs.  If you change it to the synthetic view, at least you get a nice rendered view that also makes more sense with the HUD.   It's just a nicer immediate UX, as it stands.

SamuelDudley, 2017-09-27:
Well you have convinced me re the default view change :) I'm experiencing some interesting map texture issues with the latest cesium version when terrain is enabled (always enabled in MAVCesium). Chrome still seems to work fine...
There is one other thing and that's the time of day (and hence the lighting) is driven by your local computer time. I.e testing at night will result in a darker map, while testing during the day results in light and appropriate shadows. You can turn off dynamic lighting to remove this effect. 
I probably need to review that 'feature', but I was using it to get an idea of where the sun would be during a mission e.g. am I going to be flying into the sun and loosing track of the aircraft.
Can you please post a screen capture of the 'blobs'?
Thanks!

fnoop, 2017-09-28:
Aaah, right, it's the dynamic lighting thing.  For some reason it thinks it's nighttime (1.30pm here), I think it always think it's night time when I look at it (maybe set to aussie time somehow?).   When I turn that off it looks OK.  It's very confusing, maybe turn the dynamic lighting off by default?

SamuelDudley, 2017-09-28:
The time controls the sun position relative to the earth rather than the simulation location. If you ran a simulation of a aircraft in a Europe location right now I would expect sun / light. However, because your running a simulation in Australia and the sun is over your side of the earth its dark :) I'll disable the dynamic lighting by default for you tomorrow. Cheers!

fnoop, 2017-09-28:
Oh yeah.  Duh!!!

SamuelDudley, 2017-09-29:
Dynamic lighting is now off by default https://github.com/SamuelDudley/MAVCesium/commit/eb4fd3bb1601f147d97e9bf25679bc8fa83830bc

Let me know if that works for you.
Cheers!

fnoop, 2017-09-29:
I can see the light now!  Note that when I load the page now I get a lot of artifacts, then it takes a few seconds to settle and looks OK.  Don't know if it's related to this or not.
[image]

SamuelDudley, 2017-09-29:
Because there is no relationship between the cesium terrain DEM and the DEM used in the SITL, its quite possible for the aircraft to be underground (which cesium does not handle too well). Ground interaction is always going to be a bit crappy and was one of the reasons why I was starting with the top down view (hides the artifacts). In general I find that once the DEM loads in the view settles down and it looks okay. 
There are a few options to make the ground interactions better such as:

- Not switching to this view until the terrain is fully loaded

- Clamping the aircraft to the cesium DEM (violates SITL DEM / real life)

fnoop, 2017-09-30:
Ah right (he says pretending to know what a DEM is).  Maybe top down is better default then, the main reason I didn't like it was because it was always dark, which is now fixed :)

SamuelDudley, 2017-09-30:
Sorry for the jargon! DEM = digital elevation map. Basically cesium uses a different data source to the SITL, both of which are approximations of the local terrain height above the WGS84 ellipsoid model. So if you were using this as a display and your drone went lower than the terrain model in cesium it would end up 'underground'. The ground interactions are always going to be a weak point...

SamuelDudley, 2017-10-17:
@fnoop I have rolled back to making the top down view default for the time being. I have however set the camera altitude much higher to provide a similar initial view to the MAVProxy map module. If you have a chance to view the change I would appreciate your thoughts. Cheers.

closed 2017-10-31 by SamuelDudley

fnoop, 2017-11-02:
Yes topdown looks better.  It's a little confusing as the altitude is 0.1 and you look down from quite a height, but when you first load mavcesium and look at the other camera views they're all squiffy so this is probably the best.  If I choose the 'Free' view I get a very blurry view of something, if I choose forward you get a very corrupted view as screenshot above with the black yellow and red, and if I choose 'Mount' I get a flickering tearing screen.  So the topdown is the best of the bunch until after you take off :)

SamuelDudley, 2017-11-02:
Thanks for testing! I hope that one day I can sort out the camera interactions to avoid the issues you are experiencing but until that time the top down view is the safest option.