goodrobots/MAVCesium #19: Can't restart cesium

fnoop, 2017-05-14:
Can't unload and load cesium module from mavproxy.

```
dule reload cesium
MANUAL> WebSocket connection closed: connection was closed uncleanly (peer dropped the TCP connection without previous WebSocket closing handshake)
Unloaded module cesium
MAVCesium display loaded at http://192.168.1.175:6790/
Reloaded module cesium
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1193, in run
    self.startRunning(installSignalHandlers=installSignalHandlers)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1173, in startRunning
    ReactorBase.startRunning(self)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 684, in startRunning
    raise error.ReactorNotRestartable()
ReactorNotRestartable
```

```
MANUAL> module unload cesium
MANUAL> Unloaded module cesium

MANUAL> module load cesium
MANUAL> ERROR in command ['load', 'cesium']: Couldn't listen on 192.168.1.175:6790: [Errno 98] Address already in use.
```

SamuelDudley, 2017-05-14:
This is an issue with the twisted "reactor". Once stopped it can't simply be restarted... I have been requested to port the project to tornado which should hopefully resolve this issue at the same time.

SamuelDudley, 2017-09-24:
WIP opened with initial port. #24 Some small fixes needed, but very close to replacing the existing flask + twisted arrangement.

SamuelDudley, 2017-09-26:
Fixed with #24

closed 2017-09-26 by SamuelDudley

fnoop, 2017-09-26:
Works great, thanks!