goodrobots/MAVCesium #20: Websocket address logic faulty

fnoop, 2017-05-14:
The new config parsing in master sets websocket address to the server bind address+port.  The default bind address is 0.0.0.0, which is a good thing.  However this then makes the browser attempt to connect to websocket over ws://0.0.0.0:5000/ws which is bogus.
```
WEBSOCKET = "ws://"+SERVER_INTERFACE+":"+SERVER_PORT+"/ws"
```

Perhaps a better method might be something like this:
```
        var websocket = '{{ websocket }}';
        if (/0.0.0.0/.test(websocket) || /localhost/.test(websocket)) {
          websocket = ("ws://" + window.location.hostname + ":" + window.location.port + "/ws");
        }
```
Or perhaps even better would be to make WEBSOCKET an optional configuration item and use it if set, otherwise by default construct the websocket address:
```
        var websocket = '{{ websocket }}';
        if (!websocket) {
          websocket = ("ws://" + window.location.hostname + ":" + window.location.port + "/ws");
        }
```

SamuelDudley, 2017-05-14:
Thanks for the issue and the suggestions! This needs fixing and should not be too difficult.

SamuelDudley, 2017-09-24:
Hi,
I feel I have fixed this issue with the latest commit to #24 https://github.com/SamuelDudley/MAVCesium/pull/24/commits/47b13b1f68b27a7ba4adaeefdda07b85fc792a12

If you have a chance to test it, let me know how it preforms for you.
Thanks,
Sam

SamuelDudley, 2017-09-26:
Fixed with #24

closed 2017-09-26 by SamuelDudley

fnoop, 2017-09-26:
Works well, thanks :)