goodrobots/MAVCesium #21: Configurable config?

fnoop, 2017-05-14:
New config parsing master code loads config (and api key) from mavproxy directory.  This has two problems:
 - Config shouldn't live in python mavproxy path - this will usually be owned by root and normal users won't have write access. (Default is /usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_cesium/app/).
 - Doesn't allow for multiple instances of mavproxy.  There can be any number of instances of mavproxy running (one for flight controller, one for sitl, for example).  Currently this will try and load cesium on the same port for both instances, causing a conflict and error in the second loaded instance.

SamuelDudley, 2017-05-14:
The default config is the last option which is used when all other options have failed. By default it will look for config files in the following order:
1. `_conf_file_path = os.path.join(os.environ['HOME'], ".mavcesium.ini")`
2. _conf_file_path = os.path.join(os.environ['LOCALAPPDATA'], "MAVProxy", "mavcesium.ini")
3. If the above 2 fail it will load the default config `mavcesium_default.ini` from the MAVProxy directory

Try adding a custom config file called `.mavcesium.ini` to your users home directory. It should be used over the default. This behaviour does not fix the issue of running multiple instances. To support this a config file path will need to be supplied at launch.

SamuelDudley, 2017-09-29:
Having a bit more of a think about this... I could make a two step process to start the server from MAVProxy. 

1.  Import the module. loads default config but does not start the server.
2.  User enters optional config path mavcesium load path_to_config
3.  User runs the server using the new config file. mavcesium start 

Any thoughts?

fnoop, 2017-09-29:
I think that sounds a bit complex, also won't allow auto-starting.  Is there any way of passing an extra argument to mavproxy, that can then be used by mavcesium?  That way you could specify the desired config to mavproxy at start.

SamuelDudley, 2017-09-29:
I dont think that functionality exists at the moment but would be pretty simple to add. I'll work on a MAVProxy PR to add it.

SamuelDudley, 2017-09-30:
@fnoop I would be grateful if you could **+1** the MAVProxy PR: https://github.com/ArduPilot/MAVProxy/pull/452

fnoop, 2017-12-05:
Now that mavcesium is standalone, could a specific config file location parameter be added?

SamuelDudley, 2017-12-05:
I have pushed some changes to #35 which allows the application to be ran with `--configuration="/somepath/config.ini"` . This should solve your problem :)

fnoop, 2017-12-06:
Works great, thanks :)

closed 2017-12-06 by fnoop