goodrobots/MAVCesium #23: PX4 Support

see3peoh, 2017-07-04:
How can we get this working with PX4? Would be good to have it on both stacks and the majority of the mav messages are common. Happy to help.

SamuelDudley, 2017-07-05:
Hey,
at the moment the application is driven from MAVProxy so that would be the first entry point for compatibility. After that its just a matter of supporting the right messages (if they are not the same).

I have never used the PX4 stack, but I suppose I could flash it from QGC onto a Pixhawk... Additionally I have considered making a stand alone release which does not require MAVProxy to run... Would this be of interest?

see3peoh, 2017-07-05:
Noted on the mavproxy entry point. Was looking at a few ways to bridge mav data over but nothing working yet. Will keep going through it. 

Stand alone would be epic. Then really not relying on anything and people should use this. It's awesome. 

I migrated to px4 mainly bec I was able to get lidar autoland so solid but a qgc flash and taking a look would great to know if you think it's easily feasible. 

People really should use this, it's so much better than anything else and quite a lot of ideas come to mind with how this could be additionally useful.

SamuelDudley, 2017-07-05:
Glad you like the concept! I really have been meaning to devote some more love to this project...
If your keen to share your ideas please add them here: https://github.com/SamuelDudley/MAVCesium/issues/6 :+1: 

I have been toying with wrapping the web application in a QT based gui to make it a stand alone 'desktop app' while still having the option of running it in a browser... many different ways to attack the stand alone configuration, but choosing one that will work for most cases is tricky!