goodrobots/MAVCesium #26: Make application independent of (but still compatible with) MAVProxy

SamuelDudley, 2017-09-27:
Currently MAVProxy is used to generate the JSON messages which drive the display. MAVProxy is also used for communicating back to the autopilot, WP and fence management.

Ideally this application could be made stand alone to interact with the autopilot directly.

SamuelDudley, 2017-09-29:
This would help support a wider range users including those using PX4 #23

SamuelDudley, 2017-09-29:
One idea is to have a TCP / UDP connection come into the server, parse the MAVLink there and then push the parsed data over the websocket. If I make use of python3 the process can be asynchronous.

ryanjAA, 2017-09-30:
This is really great! Standalone and/or PX4 support would open up a lot of ideas with this. Great work!

SamuelDudley, 2017-10-16:
I have pushed a minimal stand alone functionality with https://github.com/SamuelDudley/MAVCesium/commit/f0c946c58b0f1ef9b382315edd1ac3a6c7525b50 .
By running `/app/cesium_web_server.py` directly you can launch the MAVCesium server and consume a MAVLink stream independent of MAVProxy.

brizey02, 2017-11-07:
Failed to use config file at /home/flight/MAVCesium/app/mavcesium_default.ini : ConfigParser instance has no attribute 'read_file'
Traceback (most recent call last):
  File "cesium_web_server.py", line 13, in <module>
    from config import SERVER_INTERFACE, SERVER_PORT, APP_SECRET_KEY, WEBSOCKET, BING_API_KEY, APP_DEBUG, APP_PREFIX
ImportError: cannot import name SERVER_INTERFACE
Latest configparser on 14.04

SamuelDudley, 2017-11-08:
Thanks for testing. I'll see if I can reproduce. Did you clone the repo directly or install with MAVProxy?
Thanks!

brizey02, 2017-11-08:
Figured it out. Meant to remove it sorry. It's working like a charm!

SamuelDudley, 2017-11-08:
Great! Connected to software in the loop? Basically there are two ways forward to extend the stand alone functionality:

1. Expand the server code to handle the mavlink processing (basically what MAVProxy already does)
2. Process the incoming MAVLink stream in the browser directly

Because 1 already exists in some form I'm going to push on with option 2 and see how hard it it to build a GCS in a browser...

brizey02, 2017-11-08:
Yea, actually forwarded from my PC to the server. Option 2 sounds good. I was looking at a way to build a clone of dji's flighthub today for mavlink based craft but it less advanced then mavcesium! they seem to support 4 craft. Did u ever get to work on the video stuff?

fnoop, 2017-11-30:
This is great, it would be really nice to take a config file as an argument, so we can launch multiple copies.

SamuelDudley, 2017-11-30:
For multi-aircraft viewing? By forgoing the single drone limitation of MAVProxy it would be pretty simple to add any number of aircraft into the MAVCesium world to view at the same time. Doing one connection per aircraft would be pretty simple, or a single connection and change the drone model depending on the system ID.

fnoop, 2017-11-30:
No, so you can run multiple copies of mavcesium on the same computer, ie one for SITL and one for the flight controller.  If that can be done within a single instance of mavcesium, cool.

SamuelDudley, 2017-11-30:
Understood. In that case you would probably want the two separate instances of the server rather than the shared world I was getting at.
I appreciate the ideas!

ludviglee, 2018-07-09:
@SamuelDudley Amazing work！ But how can I run it in offline mode?