goodrobots/MAVCesium #27: Add SERVER_PREFIX config option to allow reverse proxy

fnoop, 2017-09-28:
Running mavcesium behind a reverse proxy is much easier if it runs on a configurable prefix (eg. /mavcesium).

SamuelDudley, 2017-09-29:
I have added a configurable prefix with [this commit](https://github.com/SamuelDudley/MAVCesium/commit/ae1216f94ae07faeb9ce669c2a888812db7cfc51). Let me know if that solves your problem or you need further changes.
Cheers!

fnoop, 2017-09-29:
All the static content is still served out of /static, that will need to move relative to the prefix as well I think?

SamuelDudley, 2017-09-29:
I think you are correct... Sorry, very green to making a web app somewhat deployable :) It shouldn't be too difficult to fix. Anything to help package this for maverick is a step in the right direction!

SamuelDudley, 2017-09-30:
With #29 applied the static files are being served correctly, but I don't know if the relative paths scattered in the javascript are going to be an issue? e.g. https://github.com/SamuelDudley/MAVCesium/blob/master/app/static/DST/js/core.js#L475
I dont think they will be an issue but I'd like to be sure from your end.

fnoop, 2017-09-30:
All the absolute pathnames in the DST/js files will need to be made relative.  How are they generated, or did you construct them by hand?

SamuelDudley, 2017-09-30:
Added by hand :(... I'll pass a javascript variable into the index.html template from the tornado server and prefix the existing file paths with it.

fnoop, 2017-09-30:
Can you not just leave relative paths in?

SamuelDudley, 2017-09-30:
The issue was that the relative paths weren't including the prefix... Should be fixed now

fnoop, 2017-09-30:
Still a few in DST/css/third_party/font-awesome.min.css and DST/js/third_party/jquery-ui.min.js

SamuelDudley, 2017-10-01:
Hey,
Not sure where you are referring to sorry. Can you link to the line in the file?
If your talking about the files in index.html:

```
    <link rel="stylesheet" type="text/css" href="{{static_url("DST/css/cesium.css")}}"/>
    <link rel="stylesheet" type="text/css" href="{{static_url("DST/css/third_party/bootstrap.min.css")}}"/>
    <link rel="stylesheet" type="text/css" href="{{static_url("DST/css/third_party/font-awesome.min.css")}}"/>
    <link rel="stylesheet" type="text/css" href="{{static_url("DST/css/third_party/jquery-ui.min.css")}}"/>
    <link rel="stylesheet" type="text/css" href="{{static_url("DST/css/third_party/lobipanel.css")}}"/>
    <link rel="icon" href="{{static_url("DST/favicon-32x32.png")}}" sizes="32x32" />
    etc...
``` 
the `{{static_url()}}` is expanded according to the tornado server `static_url_prefix` which is defined [here](https://github.com/SamuelDudley/MAVCesium/blob/master/app/cesium_web_server.py#L81).

Looking at the html of the served page and setting `mavcesium/` as the `app_prefix` in the config file I see this:
```
<link rel="stylesheet" type="text/css" href="/mavcesium/static/DST/css/cesium.css?v=931c3bc4dd44e44062101d7176637443"/>
<link rel="stylesheet" type="text/css" href="/mavcesium/static/DST/css/third_party/bootstrap.min.css?v=2f624089c65f12185e79925bc5a7fc42"/>
<link rel="stylesheet" type="text/css" href="/mavcesium/static/DST/css/third_party/font-awesome.min.css?v=73477b4d03a0989f78a352e1063b1575"/>
<link rel="stylesheet" type="text/css" href="/mavcesium/static/DST/css/third_party/jquery-ui.min.css?v=6baef79004e6978af706d6ee00e3369e"/>
<link rel="stylesheet" type="text/css" href="/mavcesium/static/DST/css/third_party/lobipanel.css?v=8aae7746e33082ffad55f1cb119bf94d"/>
<link rel="icon" href="/mavcesium/static/DST/favicon-32x32.png?v=dd662c8685816cc43cc981f9737c964b" sizes="32x32" />

```

Having a look at the sources tab in chrome shows that all of the served files have the correct prefix:
[image]

One gotcha is not clearing the browser cache, because it will continue to load the old index.html (with the old/no static prefix) 

So as far as I can tell it is working as expected with my setup... Anyway, if you can provide a link to the line where is issue is on your end that would be a big help :)

Let me know how you go.
Cheers

fnoop, 2017-10-01:
Do `grep -r '/static` and you'll see where they are.  If you open the console rather than the source in the browser devtools you can see the big red 404s :)

fnoop, 2017-10-01:
Also getting a 403 when trying to proxy the websocket:
[image]

Corresponding warnings in the mavproxy console:
```
WARNING:tornado.access:403 GET /mavlink/mavcesium/websocket/ (127.0.0.1) 0.76ms
WARNING:tornado.access:403 GET /mavlink/mavcesium/websocket/ (127.0.0.1) 0.44ms
WARNING:tornado.access:403 GET /mavlink/mavcesium/websocket/ (127.0.0.1) 0.65ms
```

fnoop, 2017-10-01:
Turning off the origin check in tornado 4 fixes the 403 error (overriding the check_origin method):
https://stackoverflow.com/a/25071488

SamuelDudley, 2017-10-01:
Thanks for your efforts with this. I'm not sure how the `/static` has ended up in the .min files but I have taken the opportunity to move to bootstrap 4, the latest jquery + font awesome.
I've almost completed the port...

SamuelDudley, 2017-10-01:
@fnoop I have pushed a bunch of changes including the websocket issue work around. `grep -r /static` is now only giving me `readme.md` so hopefully all the static paths are finally fixed!

fnoop, 2017-10-01:
Looking much better!  Few errors still in the console and the bootstrap head row doesn't collapse well:
[image]

SamuelDudley, 2017-10-01:
Yeah saw that error while testing it seems consistent so should be easy to track down... Will work on the bootstrap headders but mainly glad it's looking better from your end!

fnoop, 2017-10-01:
Ok all working now under reverse proxy.  Thanks so much for the quick response :)

closed 2017-10-01 by fnoop

SamuelDudley, 2017-10-02:
Not a problem. Glad to actually do some collaboration to improve the ArduPilot / companion computer eco system.