goodrobots/MAVCesium #28: app/mavcesium_default.ini should be in .gitignore

fnoop, 2017-09-28:
.ini should be .gitignored from commits

SamuelDudley, 2017-09-29:
Hopefully [this commit](https://github.com/SamuelDudley/MAVCesium/commit/8e1c100edc0600450e399ca0a3ec6e169787a97a) solves your issue. Let me know if it doesn't.

fnoop, 2017-09-29:
Fresh clone, and then edit of ini file:
```
[dev] [mav@maverick-ubuntuvm /usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_cesium]$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   app/mavcesium_default.ini

no changes added to commit (use "git add" and/or "git commit -a")
```

SamuelDudley, 2017-09-29:
Okay, still broken....

SamuelDudley, 2017-09-30:
My git foo is weak but because the file is already being tracked .gitignore doesn't work. If I remove the file from being tracked users won't get a copy when they clone the repo. Apparently the correct way to do this (going off the git book) is to put the config in a submodule...
At the moment I have the server looking for config files in the user's home directory and the MAVProxy app data directory and prioritising them over the example config. Will it work for you to run a config file external to the repo? 
Another fix is passing the config parh at runtime as you suggested. Alternatively you can run git update-index --assume-unchanged app/mavcesium_default.ini which should stop tracking the file in your local repo. The problem with this method is that it only works at a local repo level.

fnoop, 2017-09-30:
Yes I think you're right - this ini file should be considered a default and not to be altered.  If any alterations need to be made, the file should be copied into the home directory or the app data directory, or ideally to a specified location by mavproxy.  Can mavproxy really not call a module with a parameter?

closed 2017-09-30 by fnoop