goodrobots/MAVCesium #32: Possible to run SITL

furioushusky, 2017-11-25:
Hi, are is there a way to load cesium module from MAVProxy running with SITL?

SamuelDudley, 2017-11-25:
Hi,
Yep there sure is :) As an example the server at www.mavcesium.io is running a SITL instance connected to MAVProxy (which is running the cesium module).
If you have a SITL session running and MAVProxy is connected you can simply load the cesium module with `module load cesium` and everything should work for you.

Let me know if you run into any issues and I'll do my best to help out.

furioushusky, 2017-11-25:
Oh thanks, I actually didn't load the cesium like that but somehow I manage to load it using python command and load the cesium_web_server.py server (because I cannot load the cesium module in MAVproxy). Also I noticed that It is just a drone floating in space without the globe.

SamuelDudley, 2017-11-27:
If you are running MAVProxy I suggest loading the application via that method (rather than running cesium_web_server.py directly). Ensure you have an internet connection to be able to download tiles otherwise you will experience the black globe as per your comment above.

furioushusky, 2017-11-27:
how can I change the model of the drone? I have a DJI s1000 3D model to replace the single rotor drone shown here

SamuelDudley, 2017-12-07:
Hi,
you will either need to replace the current model files here (with a new set of files that have the same name): https://github.com/SamuelDudley/MAVCesium/tree/master/app/static/DST/models
or add a new set of models to that directory and update the vehicle model call here: https://github.com/SamuelDudley/MAVCesium/blob/master/app/static/DST/js/core.js#L12
Thanks

closed 2017-12-07 by SamuelDudley