goodrobots/MAVCesium #33: Editable HTML and JS files

furioushusky, 2017-11-27:
After successfully loaded this app, I found out that after making changes to the HTML file and other JS files, it does not reflect when running the app. Is it that it is not possible to make changes or modify the HTML and JS files?

furioushusky, 2017-11-27:
The changes I try to make is to replace the model with my animated drone model and also other stuffs in the cesium js file under DST

SamuelDudley, 2017-11-27:
Hi, there is nothing special with the HTML or js code. Perhaps your browser cache needs clearing? Try reloading the python webserver and when you have the app active in the browser push f5 or your browser cache clearing equivalent keystroke.

SamuelDudley, 2017-11-27:
Would you mind sharing your animated drone model? That way I could include it as a selection option from withing the web application.

furioushusky, 2017-11-27:
I already did the cache clearing, but the drone model does not change and other texts displayed.
I try to view the page source and it seems that I have not made any changes.

furioushusky, 2017-11-27:
However when I double click the HTML file (index), I can see the changes in the texts but running the app It shows the previous unedited one

SamuelDudley, 2017-11-27:
Are you really sure that you don't have another copy that is pulling files from else where? Apart for that I'm afraid I can't help you debug your development environment, sorry.

closed 2017-11-27 by SamuelDudley

furioushusky, 2017-12-01:
Hi, I did some observation on where does the sever picking up the sources.
Since the server set-up is a local one, should it get all the sources from my local computer?
I see that it picks it up from a (http://127.0.0.1:5000/mavcesium/ ) "mavcesium" folder. But in the mavproxy_cesium file, I dont see any file that has this name. Therefore I have no idea where does all the sources coming from

furioushusky, 2017-12-01:
Oh never mind I know where the sources comes from. Sorry to bother you