goodrobots/MAVCesium #34: Standalone mavcesium uses a lot of resource

fnoop, 2017-11-30:
```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
14994 mav       20   0  427996 112960  12180 R  20.1  6.6  26:34.26 python
```
112Mb resident memory is a lot for a small embedded computer (for example an rpi zero only has 384mb after graphics).  Can this be reduced?

The cpu is presumably down to pymavlink?

SamuelDudley, 2017-11-30:
I have yet to profile anything, but I'm sure we can save some of those resources :)
As a first step you could try increasing the blocking wait time in the select call to minimize any CPU spinning that may be occurring. Try increasing `0.01` to `1.0` and see if it reduces the overall CPU load a bit
https://github.com/SamuelDudley/MAVCesium/blob/master/app/cesium_web_server.py#L172
change to
```inputready,outputready,exceptready = select.select([self.connection.control_connection.port],[],[],1.0)```

Unfortunately (as you know) pyMAVLink is a bit of a CPU hog. But knowing where the memory and CPU is being used is a good start!

fnoop, 2017-12-01:
```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
14994 mav       20   0 1041352 723284   5112 S  20.6 42.4 275:22.41 python
```
Ouch, that's 724Mb resident after it's been running for a while.  Restarting it goes back to low memory usage (40mb) but climbs slowly.  Must either be a memory leak, or something is creating resources and releasing previous ones.

fnoop, 2017-12-01:
Changing the wait time as suggested doesn't seem to make any difference.

SamuelDudley, 2017-12-01:
Nasty! Thanks for reporting this. I'm sure I'll be able to track something down...

SamuelDudley, 2017-12-01:
I haven't had a chance to profile the code yet, but looking over it there were a couple of places that needed to be fixed. One being a infinite sized queue with no consumer! =S  
The WIP branch with the fixes is here https://github.com/SamuelDudley/MAVCesium/tree/memory_fix
#35

SamuelDudley, 2017-12-04:
Spent pretty much all day on this with no real luck....

**Some things to note:**
  - Main CPU usage is receiving MAVLink messages
  - PyMAVLink does use a fair slab of memory but remains consistent once all message types have been received (each connection stores the latest instance of each message type).
  - The real memory issues seem to be with tornado. I'm using the latest from pip `tornado-4.5.2`

**Using guppy to try and profile the memory usage:**

launching with no websocket connections:
```
Partition of a set of 843 objects. Total size = 161264 bytes.
 Index  Count   %     Size   % Cumulative  % Kind (class / dict of class)
     0     31   4    32488  20     32488  20 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_header
     1     28   3    11680   7     44168  27 dict (no owner)
     2    111  13     9768   6     53936  33 __builtin__.weakref
     3     62   7     5512   3     59448  37 array.array
     4      7   1     3408   2     62856  39 types.FrameType
     5      1   0     3352   2     66208  41 dict of pymavlink.dialects.v20.ardupilotmega.MAVLink
     6      1   0     3352   2     69560  43 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_camera_fee
                                             dback_message
     7      1   0     3352   2     72912  45 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_rc_channel
                                             s_message
     8      1   0     3352   2     76264  47 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_sensor_off
                                             sets_message
     9      1   0     3352   2     79616  49 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_servo_outp
                                             ut_raw_message
<127 more rows. Type e.g. '_.more' to view.>
```
Pretty stable around 161264 bytes used... 

Adding a connection:
```
websocket opened!
Partition of a set of 2160 objects. Total size = 433728 bytes.
 Index  Count   %     Size   % Cumulative  % Kind (class / dict of class)
     0    281  13    55440  13     55440  13 str
     1     33   2    34584   8     90024  21 dict of tornado.concurrent.Future
     2     31   1    32488   7    122512  28 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_header
     3      7   0    23464   5    145976  34 dict of tornado.iostream.IOStream
     4     49   2    21400   5    167376  39 dict (no owner)
     5    121   6    15112   3    182488  42 list
     6     13   1    13624   3    196112  45 dict of tornado.gen.Runner
     7    179   8    13288   3    209400  48 tuple
     8    149   7    13112   3    222512  51 __builtin__.weakref
     9    100   5    12000   3    234512  54 function
<179 more rows. Type e.g. '_.more' to view.>

Partition of a set of 2182 objects. Total size = 435728 bytes.
 Index  Count   %     Size   % Cumulative  % Kind (class / dict of class)
     0    281  13    55440  13     55440  13 str
     1     33   2    34584   8     90024  21 dict of tornado.concurrent.Future
     2     31   1    32488   7    122512  28 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_header
     3      7   0    23464   5    145976  34 dict of tornado.iostream.IOStream
     4     49   2    21400   5    167376  38 dict (no owner)
     5    172   8    15136   3    182512  42 __builtin__.weakref
     6    121   6    15112   3    197624  45 list
     7     13   1    13624   3    211248  48 dict of tornado.gen.Runner
     8    179   8    13288   3    224536  52 tuple
     9    100   5    12000   3    236536  54 function
<179 more rows. Type e.g. '_.more' to view.>
```
Memory usage instantly increased to 435728 bytes...
[... 29 lines cut ...]
     7    187   5    13856   2    419568  65 tuple
     8     13   0    13624   2    433192  67 dict of tornado.gen.Runner
     9    102   3    12240   2    445432  69 function
<181 more rows. Type e.g. '_.more' to view.> 
```
Memory usage has increased again while serving no more connections...
I then opened and closed a few more connections and the new memory usage while not serving any clients is 685680 bytes:
```
Partition of a set of 4917 objects. Total size = 685680 bytes.
 Index  Count   %     Size   % Cumulative  % Kind (class / dict of class)
     0   3299  67   286120  42    286120  42 str
     1     39   1   115752  17    401872  59 dict (no owner)
     2    471  10    84280  12    486152  71 list
     3     31   1    32488   5    518640  76 dict of
                                             pymavlink.dialects.v20.ardupilotmega.MAVLink_header
     4    189   4    16632   2    535272  78 __builtin__.weakref
     5     27   1     7560   1    542832  79 dict of tornado.template._Text
     6     31   1     7192   1    550024  80 __builtin__.set
     7     83   2     6424   1    556448  81 tuple
     8     20   0     5600   1    562048  82 dict of tornado.template._Expression
     9     62   1     5512   1    567560  83 array.array
<149 more rows. Type e.g. '_.more' to view.>
```
So clearly something within the tornado server is not cleaning up after itself...
Possibly related https://stackoverflow.com/questions/47486897/tornado-websocket-server-doesnt-release-ram-memory

SamuelDudley, 2017-12-04:
with regards to CPU usage we can see that most of the time spent by pyMAVLink is parsing the incoming char buffer 

```
Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
   314                                               def recv_msg_wrapper(self):
   315                                           #         lp_wrapper = lp(self.process_connection_in)
   316                                                   '''message receive routine'''
   317       729         3252      4.5      0.8          self.pre_message()
   318      1458         1683      1.2      0.4          while True:
   319      1458        13895      9.5      3.5              n = self.mav.bytes_needed()
   320      1458        11888      8.2      3.0              s = self.recv(n)
   321      1458         1698      1.2      0.4              numnew = len(s)
   322                                           
   323      1458         1295      0.9      0.3              if numnew != 0:
   324      1458        10525      7.2      2.7                  if self.logfile_raw:
   325                                                               self.logfile_raw.write(str(s))
   326      1458         1397      1.0      0.4                  if self.first_byte:
   327         1           10     10.0      0.0                      self.auto_mavlink_version(s)
   328                                           
   329                                                       # We always call parse_char even if the new string is empty, because the existing message buf might already have some valid packet
   330                                                       # we can extract
   331      1458       283133    194.2     71.6              msg = self.mav.parse_char(s)
   332      1458         6292      4.3      1.6              if msg:
   333       729          800      1.1      0.2                  if self.logfile and  msg.get_type() != 'BAD_DATA' :
   334                                                               usec = int(time.time() * 1.0e6) & ~3
   335                                                               self.logfile.write(str(struct.pack('>Q', usec) + msg.get_msgbuf()))
   336       729        58211     79.9     14.7                  self.post_message(msg)
   337       729          877      1.2      0.2                  return msg
   338                                                       else:
   339                                                           # if we failed to parse any messages _and_ no new bytes arrived, return immediately so the client has the option to
   340                                                           # timeout
   341       729          677      0.9      0.2                  if numnew == 0:
   342                                                               return None
```
`self.mav.parse_char(s)` is within the generated dialect code with options for speedup via mavnative but note that mavnative isn't supported for MAVLink2 yet...

```
Line #      Hits         Time  Per Hit   % Time  Line Contents
==============================================================
 10485                                                   def parse_char_wrapper(self, c):
 10486                                                       '''input some data bytes, possibly returning a new message'''
 10487       429         1423      3.3      1.8              self.buf.extend(c)
 10488                                           
 10489       429          856      2.0      1.1              self.total_bytes_received += len(c)
 10490                                           
 10491       429          370      0.9      0.5              if self.native:
 10492                                                           if native_testing:
 10493                                                               self.test_buf.extend(c)
 10494                                                               m = self.__parse_char_native(self.test_buf)
 10495                                                               m2 = self.__parse_char_legacy()
 10496                                                               if m2 != m:
 10497                                                                   print("Native: %s\nLegacy: %s\n" % (m, m2))
 10498                                                                   raise Exception('Native vs. Legacy mismatch')
 10499                                                           else:
 10500                                                               m = self.__parse_char_native(self.buf)
 10501                                                       else:
 10502       429        73533    171.4     91.8                  m = self.__parse_char_legacy()
 10503                                           
 10504       429         1724      4.0      2.2              if m != None:
 10505       214          287      1.3      0.4                  self.total_packets_received += 1
 10506       214         1288      6.0      1.6                  self.__callbacks(m)
 10507                                                       else:
 10508                                                           # XXX The idea here is if we've read something and there's nothing left in
 10509                                                           # the buffer, reset it to 0 which frees the memory
[... cut ...]
 10754       136          357      2.6      0.5                  m._payload = msgbuf[6:-(2+signature_len)]
 10755       136          240      1.8      0.3                  m._crc = crc
 10756       136          928      6.8      1.3                  m._header = MAVLink_header(msgId, incompat_flags, compat_flags, mlen, seq, srcSystem, srcComponent)
 10757       136          196      1.4      0.3                  return m
```
There you have it... So if you have ever wondered where pyMAVLink spends most of its CPU time, wonder no more!

fnoop, 2017-12-05:
re: cpu that's a really good target for improvement, for pymavlink!  Worth posting as an issue there?
Agree as suspected it has nothing to do with mavcesium

fnoop, 2017-12-05:
So here is mavcesium memory profiled with no clients attached:
[image]
The interesting thing to note is the stepped increase in program data memory (and therefore VM memory) where the step time increases over time, almost like a backoff algorithm.  Restarting it shows the same thing start again:
[image]
Now profiling again with the memory_fix branch.

SamuelDudley, 2017-12-05:
At the moment there aren't any changes that I would expect to make a difference in the memory_fix branch... Only a few small changes but nothing to fix what you are experiencing... It's interesting to note the memory climb without any clients connected, in my testing I didn't see anything nearly as pronounced as the graph you have.
What time scale is that graph over minutes or hours?

SamuelDudley, 2017-12-05:
My first thought was that the mavlink connection was keeping some sort of memory based log of received messages, causing the increase usage over time. I did go digging and didn't find any suspicious code... Not to say the issue isn't with pymavlink. I just didn't find it while looking for it.

fnoop, 2017-12-05:
Timescale is along the bottom, first graph is about 8 hours, second is over 6 hours.  Interestingly since restarting again from the memory_fix branch it uses less CPU:
[image]

I'll do a couple more tests - one without the nginx reverse proxy and another one against a very simple dronekit app, to see if the problem exists with just pymavlink.

SamuelDudley, 2017-12-05:
The majority of the CPU usage reduction comes with the reduction in message stream rate... less messages to parse means less CPU used. The issue is that the display becomes a bit 'blocky' with updates. We can tune the message rate to help balance the CPU usage if that would help (or make the rate a launch option?)

fnoop, 2017-12-05:
Ah OK.  Memory usage of mavlink-params, which is a very simple dronekit app to manage parameters, has a completely static memory usage.  So the problem isn't inherently present in pymavlink.

SamuelDudley, 2017-12-05:
Thanks for testing. I'm pretty sure it's related to tornado but just need to pin down where...

SamuelDudley, 2017-12-06:
right... so I'm pretty sure its in pyMAVLink.
by commenting out `self.connection.control_connection.recv_msg()` and replacing it with time.sleep(0.01) the memory leak is gone. Additionally if I put that bit of code back in and don't do anything with the received  message the memory leak comes back.
So somewhere in `recv_msg()` something is growing over time...

SamuelDudley, 2017-12-06:
@fnoop pretty sure I have tracked down the source of the memory leak and plugged it. Could you please test the latest commit in #35 when you have a chance...
Thanks !

fnoop, 2017-12-06:
Oh well done!
[image]

fnoop, 2017-12-06:
Is that a bug in pymavlink, or a feature?

closed 2017-12-06 by fnoop

SamuelDudley, 2017-12-06:
I'm going to call it a bug... The behavior we were seeing is the result of a bytearray being extended over and over again.
In the dialect generated code e.g. ardupilotmega.py there is a `MAVLink` object that stores the connection buffer / bytearray in `self.buf` and an associated index in `self.buf_index`.
In the `parse_char()` method of this object the problem occurs (note I have removed some unused mavnative logic below for clarity):
```
        def parse_char(self, c):
            '''input some data bytes, possibly returning a new message'''
            self.buf.extend(c)
            self.total_bytes_received += len(c)
            m = self.__parse_char_legacy()
            if m != None:
                self.total_packets_received += 1
                self.__callbacks(m)
            else:
                if self.buf_len() == 0 and self.buf_index != 0:
                    self.buf = bytearray()
                    self.buf_index = 0
            return m
```
The issue is that every time this code is run there is __always__ a complete message waiting to be decoded. Because of this the final `else` statement never gets ran and the buffer continues to grow.
Although the logic in that statement `self.buf_len() == 0 and self.buf_index != 0` is true every time this method is ran, `self.buf` and `self.buf_index` are never able to be reset... personally I think the else could be removed and we should be testing that condition every time that method is called. e.g.:
```
        def parse_char(self, c):
            '''input some data bytes, possibly returning a new message'''
            self.buf.extend(c)
            self.total_bytes_received += len(c)
            m = self.__parse_char_legacy()
            if m != None:
                self.total_packets_received += 1
                self.__callbacks(m)
            if self.buf_len() == 0 and self.buf_index != 0:
                self.buf = bytearray()
                self.buf_index = 0
            return m
```
which equivalent to what is now in the MAVCesium server code

SamuelDudley, 2017-12-06:
pyMAVLink issue raised: https://github.com/ArduPilot/pymavlink/issues/130

pietrodn, 2018-07-05:
Did you also observe increased CPU utilization over time?

PR #35 (2017-12-06): cesium_web_server - reduce memory and CPU usage
This PR will be used to fix #34 
I have not yet profiled the code, but a couple of immediate issues were present (and have been fixed).