goodrobots/MAVCesium #42: ConfigParser instance has no attribute 'read_file'

thirafi98, 2018-06-01:
Hello there, i was trying to load the cesium module but i got this error like this, there's an error say that the configparser didn't have attribute 'read_file', how can i fix this ?, this is the error line

```
Failed to use config file at C:\Python27\lib\site-packages\MAVProxy\modules\mavproxy_cesium\app\mavcesium_default.ini : ConfigParser instance has no attribute 'read_file'
ERROR in command ['load', 'cesium']: 'Configuration' object has no attribute 'MODULE_DEBUG'
Traceback (most recent call last):
  File "mavproxy.py", line 521, in process_stdin
    fn(args[1:])
  File "mavproxy.py", line 343, in cmd_module
    load_module(modname, **kwargs)
  File "mavproxy.py", line 296, in load_module
    module = m.init(mpstate, **kwargs)
  File "C:\Python27\lib\site-packages\MAVProxy\modules\mavproxy_cesium\__init__.py", line 243, in init
    return CesiumModule(mpstate, **kwargs)
  File "C:\Python27\lib\site-packages\MAVProxy\modules\mavproxy_cesium\__init__.py", line 49, in __init__
    ('debug', bool, self.config.MODULE_DEBUG)])
AttributeError: 'Configuration' object has no attribute 'MODULE_DEBUG'
```

thx

SamuelDudley, 2018-06-02:
Hi, this might be a version issue... Try upgrading configparser and let me know if that solves your issue. `pip install upgrade configparser`
Thanks.

thirafi98, 2018-06-19:
hi, sorry for waiting, while i try to upgrade configparser, there's an error like this

`
  Could not find a version that satisfies the requirement upgrade (from versions: )
No matching distribution found for upgrade`

what thing i should do next ? thx

thirafi98, 2018-06-19:
hi, sorry for wait, i've try to upgrade it, and try again but it's still the same error, what i should do next ? thx

ludviglee, 2018-07-09:
Hi,
I get the similar error when I run the cesium_web_server. And I update the configparser, It still the same error.
_Failed to use config file at D:\MyProject\MAVCesium\app\mavcesium_default.ini : ConfigParser instance has no attribute 'read_file'
Traceback (most recent call last):
  File ".\cesium_web_server.py", line 279, in <module>
    module(optsargs)
  File ".\cesium_web_server.py", line 182, in __init__
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File ".\cesium_web_server.py", line 141, in main
    server = start_app(config=config, module_callback=module_callback)
  File ".\cesium_web_server.py", line 102, in start_app
    application = Application(config, module_callback)
  File ".\cesium_web_server.py", line 86, in __init__
    (r"/"+config.APP_PREFIX, MainHandler, args),
AttributeError: 'Configuration' object has no attribute 'APP_PREFIX'

self.main_loop()
  File ".\cesium_web_server.py", line 249, in main_loop
    self.process_connection_in() # any down time (max 0.1 sec) occurs here
  File ".\cesium_web_server.py", line 215, in process_connection_in
    self.connection.control_connection.recv_msg()
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 318, in recv_msg
    s = self.recv(n)
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 989, in recv
    data = self.port.recv(n)
socket.error: [Errno 10053]_